QuestNav/QuestNav #2: Deployment and USBC-Ethernet Adapters

Torchtopher, 2024-12-01:
Have a quest 3s. 
I've set up unity to build an APK and I install that with the Meta Quest Developer Hub. Is this the correct way to deploy?
That does seems to work, and I can view the programs printouts in the log viewer. However I can't find where the motionrecorder script is writing data. Where are you able to find/read this file?  

Do you have a link to the usbc to ethernet adapter you used? I've only tried one so far but I'm not sure what to look for in them.

juchong, 2024-12-03:
Hi @Torchtopher! I just pushed a *big* update to the README that should help answer some of your questions. Be sure to check it out!

Q: Is [MQDH] the correct way to deploy?
A: Sure. I usually install the apk using `adb install app.apk`, but MQDH makes it a bit easier. 

The MotionRecorder script was the very first proof of concept app that I wrote when trying out this concept. I recommend using AdvantageKit and/or WPILib logger to record the pose data. The latest robot code example should let you easily record data onto a USB drive plugged into the RIO. 

I have a bunch of USB-Ethernet adapters on-order. I'll be sure to update the README once I test them out. Generally speaking, the more generic the adapter, the better.

Torchtopher, 2024-12-03:
New readme looks awesome! Definitely has the things I was slowed down by. 

We've managed to get a proof of concept also working with our ROS codebase, so once we can mount should be able to test some as well and hopefully integrate it with apriltags. 
Here's our version of motionstreamer if your interested [link](https://github.com/FRC900/pose-questimator/blob/main/unity/Assets/Robot/MotionStreamer.cs 
)

Also I have found this working ethernet adapter, bit expensive but it does work: https://www.aukey.com/products/cb-c71-8-in-1-usb-c-hub-with-ethernet-port

juchong, 2024-12-05:
@Torchtopher I love the name "Questimator" 😂

I added links to a few adapters I've found to work. Definitely take a look at the power supply section of the readme since it'll prevent you from running into weird disconnects. Also, be sure to pull in the improvements that were landed yesterday. I'm planning on shipping back the charge state of the headset for robot-side monitoring, as well as playing with April tag detection on-device. I found a neat C# library for Unity that's already been compiled for the tag family used on FRC fields: https://github.com/juchong/com.reachcloud.apriltag

Torchtopher, 2024-12-05:
Being able to detect tags with this seems super exciting! Also merged in the prs into my fork.

I had a few other questions. Why does the adapter need to also give power? Shouldn't continuously charging it in the pits also work? 

Thoughts about mounting the headset upside down? It seems like the cameras are mostly pointing down but if the mount is also low to the ground you would want to point up.

Our attempt at a mount isn't quite ready yet, but hopefully can get some testing done today on our robot, maybe integrating apriltags. Do you happen to have a cad model of the 3s you can share? We haven't been able to find one on the internet.

juchong, 2024-12-07:
Supplying 5V to the adapter (and headset) externally ensures that the adapter always has enough power to remain active. 

Think about it, the cameras are designed to view the world and the player's hands, controllers, etc. Mounting the headset low to the ground and/or upside down is probably not the right design tradeoff. 

I'm currently testing a mount design for the 3S. You'll be able to find it on printables sometime tomorrow morning. I don't have a Quest 3S model I can share, unfortunately. 

It looks like this workaround is probably our best shot for getting April tag support on the headset (in Unity): https://github.com/trev3d/QuestDisplayAccessDemo

Torchtopher, 2024-12-08:
All makes sense, thanks for the advice (and the mount). 

I wanted to also ask about some reliability/connection issues we've been seeing. The headset seems rock solid over wifi and can handle any disconnects etc (our code is slightly different than yours). For testing at our last meeting we actually used wifi on the robot network, which obviously needs to change. 
It seems like getting ethernet connection is more picky. Often I see a message that's like "wifi not detected please check settings" and it wont send any data. Do you have wifi off or on? I'm getting another adapter tomorrow that should definitely work with the quest + power so I'll see how that goes also. 
I was thinking it would be nice to display debug info as text in the oculus app, is there any useful debug info that can be accessed with regards to ethernet connection/status? IP address/connections maybe?

Torchtopher, 2024-12-08:
Just to add some more information, this is what I see when plugging in my adapter, and I have to click cancel, is that normal? It also seems to not work sometimes.  
[image]

juchong, 2024-12-20:
It looks like you're using a bit more than a USB-Ethernet adapter. The headset seems to think you're plugging in a storage device.

closed 2024-12-29 by juchong