QuestNav/QuestNav #9: Errors occur during build due to missing references for WebSocketSharp and related types.

NickAnastasoff, 2024-12-24:
Error Logs:
Error CS0246: WebSocketSharp not found.
File: Assets/Robot/NetworkTables-CSharp/Nt4Client.cs
Line: 7

Error CS0246: MessageEventArgs not found.
File: Assets/Robot/NetworkTables-CSharp/Nt4Client.cs
Line: 113

Error CS0246: ErrorEventArgs not found.
File: Assets/Robot/NetworkTables-CSharp/Nt4Client.cs
Line: 150

Error CS0246: CloseEventArgs not found.
File: Assets/Robot/NetworkTables-CSharp/Nt4Client.cs
Line: 155

Error CS0246: WebSocket not found.
File: Assets/Robot/NetworkTables-CSharp/Nt4Client.cs
Line: 39

Build fails: "Scripts have compile errors in the editor."
Warning: "Build completed with result 'Unknown'."

Meta > OVR Build > OVR Build APK… doesn’t show up for me so I’m using File > Build Profile > Build with development mode checked. Any help is appreciated!
[image]

ehochstein, 2024-12-24:
Check this post to solve the OVR issue: https://www.chiefdelphi.com/t/questnav-the-best-robot-pose-tracking-system-in-frc/476083/170?u=ehochstein

NickAnastasoff, 2024-12-24:
> This took about 10 minutes, and then afterwards the OVR Build option showed up under the Meta menu.
For me, when I run it, I get
`
Build fails: "Scripts have compile errors in the editor."
Warning: "Build completed with result 'Unknown'."
`
I think the bigger issue is resolving the WebSocketSharp dependency, but i'm not sure how.

ehochstein, 2024-12-24:
> > This took about 10 minutes, and then afterwards the OVR Build option showed up under the Meta menu.
> 
> 
> 
> For me, when I build it, I get
> 
> `
> 
> Build fails: "Scripts have compile errors in the editor."
> 
> Warning: "Build completed with result 'Unknown'."
> 
> `
> 
> I think the bigger issue is resolving the WebSocketSharp dependency, but i'm not sure how.

I had similar errors on my first import of the project into Unity. To solve these errors, I imported the MessagePack plugin, then closed Unity and imported a **new** QuestNav project. That cleared any WebSocket errors I had.

juchong, 2024-12-25:
Hi! Like others have mentioned, it sounds like you haven't installed the WebSocket plugin. Be sure to follow step #5: https://github.com/juchong/QuestNav?tab=readme-ov-file#step-5-install-the-messagepack-plugin-for-unity

NickAnastasoff, 2024-12-27:
All of thats imported, I tried removing it and adding it back, still no luck.

juchong, 2024-12-27:
The only other thing I can suggest is rebuilding the entire project or trying another PC.

NickAnastasoff, 2024-12-28:
Just set it up on a windows machine and it worked! Thanks!

closed 2024-12-28 by NickAnastasoff