QuestNav/QuestNav #11: Cannot Disable Gaurdian on new Build

SeanErn, 2024-12-28:
As far as I am aware, on the new Quest 3s build, I have not been able to find a way to permenently disable gaurdian. The only workaround I've found is running this adb command, but it doesn't persist between restarts.
```shell
adb shell setprop debug.oculus.guardian_pause 1
```

SeanErn, 2024-12-29:
Ok, after more looking, there is an option under advanced options to disable it, but again, it does not persist

juchong, 2024-12-29:
@SeanErn be sure to review the One-Time setup instructions. https://github.com/juchong/QuestNav?tab=readme-ov-file#one-time-headset-setup-required-for-each-headset

These settings should persist. 

- Disable the guardian for development purposes Settings > Advanced > Experimental Settings > Enable Custom Settings and TURN OFF Physical Space Features, MTP Notification, and Link Auto Connect
   - These settings might also be located in Settings > Developer > Experimental Settings > Enable Custom Settings on some older OS builds

SeanErn, 2024-12-29:
Thats so weird! On my headset after I set it in the GUI and restarted via adb, the option turned off.

SeanErn, 2024-12-29:
I am going to check the software version of my headset tomorrow. Maybe factory reset it to start with a clean slate. Thanks again!

SeanErn, 2024-12-29:
Can confirm, after resetting my headset the GUI setting persists.

closed 2024-12-29 by SeanErn