QuestNav/QuestNav #13: Rotation of robot causes pose to change

SeanErn, 2024-12-30:
Hello! This is most likely more of an issue with the way I'm implementing QuestNav than an actual issue with the program, but I wanted to see if anyone had any input. 

All of this data is made with the latest stable build (not my dev build)

When my robot rotates, each of the translations for the poses follow a sinewave-like pattern:
[image]
Blue and Yellow represent X and Y transforms of robot pose

To confirm my robot is not moving the same way, this is X transform from Oculus plotted against X transform from wheel odometry (Differentiated)
[image]
The drive wheel pose is relatively stable while the oculus pose moves significantly.

I use the following offset logic, which I've confirmed is **physically** correct.
```java
private Translation2d getOculusPosition() {
    return new Translation2d(inputs.position[2], -inputs.position[0]);
  }

  public Pose2d getOculusPose() {
    var oculusPositionCompensated = getOculusPosition();
    var pose = new Pose2d(oculusPositionCompensated, Rotation2d.fromDegrees(getOculusYaw()));
    return pose;
  }

  public Pose2d getRobotPose() {
    var oculusPose = getOculusPose();
    var robotPose = getOculusPose().transformBy(OCULUS_TO_ROBOT);
    return robotPose;
  }
```

```java
// 16cm forwards from center, centered laterally, facing robot forwards
public static final Transform2d OCULUS_TO_ROBOT =
      new Transform2d(
          0.16, 0, Rotation2d.fromDegrees(0)); // X: +Forwards, Y: + Left, ROT: + CCW 
```

At this point, I'm kind of at a loss. Any help or input, no matter how small, would be appreciated.
My full codebase is [here](https://github.com/5152Alotobots/5152_Template/blob/questnav/src/main/java/frc/alotobots/library/subsystems/vision/questnav/OculusSubsystem.java). Its stuffed with debug logging right now so it might be a little hard to read.

I can also provide comprehensive AdvantageKit logs if need be.

SeanErn, 2024-12-31:
> Is your offset measured to the front of the headset, or to the "third eye" location?

Ah. I was measuring to just behind the cameras of the headset. I'll go try to jerry-rig the offset :rofl:

juchong, 2024-12-31:
>Ah. I was measuring to just behind the cameras of the headset. I'll go try to jerry-rig the offset 🤣
You want to measure to the user-side of the fresnel lenses!

SeanErn, 2025-01-01:
> > Ah. I was measuring to just behind the cameras of the headset. I'll go try to jerry-rig the offset 🤣
> 
> You want to measure to the user side of the fresnel lenses (where your eyes go)!

I went ahead and remeasured the values, and now it works partially. As long as I transform based on the inverse value of the offset, the X value stays constant. Like this:
```java
var robotPose = getOculusPose().transformBy(OCULUS_TO_ROBOT.inverse());
```
However the Y value changes still. If I don't use the inverse, the X value changes and the Y constant. My quest should only need a X axis transform because latterly it is centered. Any ideas? We are getting closer.

SeanErn, 2025-01-01:
I'm closing this as completed as I solved the issue. I had to negate the yaw parameter. and then it worked great.

closed 2025-01-01 by SeanErn

juchong, 2025-01-02:
@SeanErn I'm not sure if this was obvious from looking at the data, but the orientation of the headset axes are different from those typically used on a robot. On a robot, Z is always yaw. However, in the Quest the Z axis points out of the cameras. 

[image]