QuestNav/QuestNav #14: Ethernet Adapter Only Works on Power Up

hamster1147, 2024-12-31:
Not sure if I'm alone in this issue, but I can only get the network adapter I bought (`Belkin Connect USB-C to Ethernet + Charge Adapter 100W`) to work when I have it plugged in and power the Quest up from cold. If I plug it in after its on, it doesn't show up. It also doesn't show up if the NIC was unplugged and plugged back in. In both cases I was not able to see it as a network device via `ifconfig` using ADB over WiFi.

I've been testing a lot of methods for ensuring there is a straight forward way to get the Quest setup for QuestNav. I'm also curious if anyone is able to have the wired network work, and also be able to talk to the device over USB ADB. I'm guessing this is a host vs client USB type of problem?

SeanErn, 2024-12-31:
> to work when I have it plugged in and power the Quest up from cold.
I also use this network adapter, but don't have this issue. Do you have WIFI completely disabled? The only issue I've seen is sometimes the adapter is flaky if I plug it in after boot. (Sometimes fails to reconnect to NT after rebooting robot code, or just drops.) So I'm unsure.

juchong, 2024-12-31:
I've found that the adapter must be powered by a 5V source (not a power source that supports/negotiates PD) before being plugged into the headset. Also, be sure to disable WiFi since it's not clear what the headset's routing table will do when two adapters are present.

SeanErn, 2024-12-31:
Ah! That probably explains my issue. I've been using an Anker power bank that does USB C PD.

SeanErn, 2024-12-31:
>  disable WiFi

Definitely, I know there is a config in adb that you can set to prefer en0, but I am not at my PC right now. When I get back I'll share it. Might be helpful just in case something is forgotten.

juchong, 2024-12-31:
>Ah! That probably explains my issue. I've been using an Anker power bank that does USB C PD.

I've found that some adapters are more tolerant of 9V (negotiated PD) than others. The Bekin adapter only seems to like 5V, but the ONN adapter ([link](https://www.walmart.com/ip/onn-8-in-1-USB-C-Adapter-USB-3-0-and-4K-HDMI-Compatible/590617670)) seems to be happy.

hamster1147, 2024-12-31:
Yup, giving power was the problem. Odd that it was able to work powering up without giving the adapter input power. I hadn't tested it with additional power yet since it was seemingly working without it at first.

hamster1147, 2024-12-31:
A discord would be pretty nice to have discussions on. I'm playing around with the ins/outs of trying to get a setup going so that our team can potentially use it this season.

closed 2024-12-31 by hamster1147