QuestNav/QuestNav #16: WebSocket

notyou05, 2025-01-05:
[image]
I am getting these error messages when I open the project. When I open it wants to go into safe mode. Also, I do have the MessagePack plugin installed. Can anyone help?

notyou05, 2025-01-05:
Also why do I have to install the example.apk and the unity project

juchong, 2025-01-05:
@notyou05 please be sure to follow all of the setup instructions in order. 

>... please add debug authorization
This error message tells me that the headset isn't in developer mode.

notyou05, 2025-01-06:
Thanks, do I still have to do all the extra work with Unity or does "QuestNav_0.4.0.apk" work?

juchong, 2025-01-06:
The .apk works. Be sure to follow the headset setup instructions closely.

notyou05, 2025-01-06:
Thanks so much are team greatly appreciated your help

notyou05, 2025-01-06:
@juchong to visualize the data in AdvantageScope do need to do any changes to are robot code or will it just work because it is already sending it to the network table?

SeanErn, 2025-01-07:
> to visualize the data in AdvantageScope do need to do any changes to are robot code or will it just work because it is already sending it to the network table?

As it is published to NT by default you should be able to get  **some** data from the headset directly. Keep in mind that you should probably log the data from the `getOculusPose()` method instead, as it properly transforms the two coordinate systems (Unity -> FRC) Just put something like this in the periodic method.
```java
Logger.recordOutput("Oculus/HeadsetPosition", getOculusPose());
```

closed 2025-01-10 by notyou05