QuestNav/QuestNav #25: Quest Connectivity Issue

MeeperTheGreat99, 2025-03-20:
The quest can't seem to connect to our robot. In the headset, the UI says "No Adapter Found", but we have it plugged into our robot's ethernet switch as well as an external 5v power supply via buck converter (Redux). The quest has wifi and bluetooth turned off, and both the Unity and Java code are up to date.

MeeperTheGreat99, 2025-03-21:
I should also add that this didn't happen before all the network code changes, this started after we updated to latest.

juchong, 2025-03-24:
Hey @MeeperTheGreat99! Give v0.8.0 a try. Be sure to add the heartbeat code to your robot project.

steinra2002, 2025-03-25:
I tried v0.8.0 tonight and noticed that even through network tables data was flowing that the status message still showed it trying to connect.  I have not had a chance to go through the code yet to see if it is maybe missing a status and updating again.  I had heartbeat added to the robot and that was flowing as it should in the network tables and incrementing.  I did drop back to the build prior that I did as it seemed more stable.  Also I noticed more calibration issues with 0.8.0 which I don't understand.

I am also confused on travel mode.  I seem to have more issues with position when travel mode is off than when on.  I know that was a comment where it may not need to be on.

I am still fighting an issue on distance traveled as well.  (probably should be a separate issue)  When traveling in a straight line it seems to be falling a little short (6+ inches) from the back of the reef to the alliance wall.

Side to side same type of thing.

Any ideas on what to try to do to validate the distances?  We set the offset from center based on the 5010 calibration routine concept, but when calculating it using the code the offset comes up as a meter+ offset, and in the positive direction when the robot placement is in the negative location.  Also I found when looking at the robot when spinning in place it drifts to the right about an 1" when turning in both directions, which is likely not helping the calibration process.

Will light levels affect position at all?

Sorry lots of items doing a late night mind dump on thoughts.

steinra2002, 2025-03-25:
I knew I forgot something.  We have seen the pose jump slightly as well.  After moving and coming to a stop a few moments later the post suddenly jumps.

We are not using a pose estimator and just looking at the Quest pose directly, with the offset translation applied.

juchong, 2025-03-25:
> I did drop back to the build prior that I did as it seemed more stable. 

Are you testing with v0.7.0 or v0.8.0?

>Also I noticed more calibration issues with 0.8.0 which I don't understand.

Can you elaborate on this a bit more? 

>I am also confused on travel mode. I seem to have more issues with position when travel mode is off than when on. I know that was a comment where it may not need to be on.

Travel mode should be *disabled* for best performance

>I am still fighting an issue on distance traveled as well. (probably should be a separate issue) When traveling in a straight line it seems to be falling a little short (6+ inches) from the back of the reef to the alliance wall. Side to side same type of thing.

Does the headset have a clear view out of your robot? This behavior seems like the headset isn't sure where it's located, which usually happens because it can't see the world around it. Are you blocking the cameras between tests? 

> Any ideas on what to try to do to validate the distances? We set the offset from center based on the 5010 calibration routine concept, but when calculating it using the code the offset comes up as a [...]

This _really_ sounds like the headset is not working correctly. Did you disassemble the headset? Does it work correctly if you put it on your head? Did you accidentally enable "privacy mode"?

steinra2002, 2025-03-25:
> > I did drop back to the build prior that I did as it seemed more stable.
> 
> Are you testing with v0.7.0 or v0.8.0?

v0.7.0 but based on the changes I did the pull request on.

> 
> > Also I noticed more calibration issues with 0.8.0 which I don't understand.
> 
> Can you elaborate on this a bit more?
>

When running 8 vs 7 I noticed it seemed to have more error in travel distance.
 
> > I am also confused on travel mode. I seem to have more issues with position when travel mode is off than when on. I know that was a comment where it may not need to be on.
> 
> Travel mode should be _disabled_ for best performance
>

This is where I see the most error, the travel in the pose does not even come close to what it does when travel mode is enabled with spatial access allowed.

We have also seen the pose grow in height (up and down at an angle in 3d pose info) and sometimes after grabbing the headset off the robot the view in the screen is at an angle.  This could affect the horizontal travel distance if the plane has shifted I guess.  Would make sense I think if that was happening.  Also sometimes exiting questnav back to the main screen it clips below the "floor".

 
> Does the headset have a clear view out of your robot? This behavior seems like the headset isn't sure where it's located, which usually happens because it can't see the world around it. Are you [...]

Clear view out of the side, we should have a full 180 degree view.  It is level in mount and using the 3d printed holder.
> This _really_ sounds like the headset is not working correctly. Did you disassemble the headset? Does it work correctly if you put it on your head? Did you accidentally enable "privacy mode"?

Using the stock headset no modifications.  Only the "ears" are rotated up.  looked at the privacy area, it looks like questnav is allowed access to spatial data, not sure what other settings to look at.  Should I try clearing the spatial data?

It does seem to work fine when playing First Encounters.  But when I just set it down and picked it back up the scene is now at a slight angle.  Rebooting the quest has seemed to correct this and the above, but I had not had this happen before just setting it down while on.  Any other resets I can try?

SeanErn, 2025-03-25:
I'm not a meta engineer, but this definitely seems like a hardware issue. Ensure your headset is up to date (v74). From v0.7 to v0.8 no logic changed at all, rather only refactoring and enabling the heartbeat system. Our team ran these changes at our last competition and experienced no issues.

Let me know if there is anything else I may be able to help with.

steinra2002, 2025-03-25:
> I'm not a meta engineer, but this definitely seems like a hardware issue. Ensure your headset is up to date (v74). From v0.7 to v0.8 no logic changed at all, rather only refactoring and enabling the [...]

Was on 71, moving to 74, will try that as well.

steinra2002, 2025-03-25:
Added to the main quest post on CD to attach screen shots of the pose.  Seems to be a pitch and yaw issue maybe?  Easy way to reset those to get to flat plane?

SeanErn, 2025-03-28:
> Added to the main quest post on CD to attach screen shots of the pose. Seems to be a pitch and yaw issue maybe? Easy way to reset those to get to flat plane?

Have you figured this out yet? Just wondering if I can close this issue.

steinra2002, 2025-03-31:
Not figured out yet, but I hijacked the connectivity issue here.  So if that is solved I can open a different issue as we look at it more in our off season.  We wrapped up comps yesterday and did not advance, so we will be now moving into a more testing side.  We will look at visibility of the cameras and placement to see if that changes things.  The flying pose seems to be an interesting issue we will dig into.

closed 2025-03-31 by SeanErn