QuestNav/QuestNav #28: AprilTag detection MVP using the passthrough API

juchong, 2025-03-24:
An AprilTag detection MVP (minimum viable product) would help users understand how tags _should_ be used, and how it differs from classical tracking approaches.

SeanErn, 2025-03-24:
I think using EmguCV would be a good full stack solution here. I am trying it now.

SeanErn, 2025-03-25:
> So is this actually "ready"? Is so that was quite quick.

Unfortunately not. This will most likely be not ready til' next season. OpenCV libraries for unity that are compatible with andriod are closed source and extremely expensive meaning that a custom compatibility layer/solvePNP implementation has to be written as OpenCV does not support .NET natively.

I am definitely working on it however. Our team needs the quest back for testing purposes so dev may slow down a bit. Don't rely on it being released this year.

crummyh, 2025-03-31:
Which branch is being used for the AprilTag enhancement? There is the `aprilTagDetection` branch, which @SeanErn seems to be working on, as well as the `passthrough` branch which @juchong is working on. I would like to help with the issue but don't know which to look at.

SeanErn, 2026-01-21:
Complete with PR #151

closed 2026-01-21 by SeanErn