QuestNav/QuestNav #31: AprilTag detection and tracking

juchong, 2025-03-24:
This is the umbrella AprilTag enhancement milestone.

blaze-developer, 2025-12-16:
Wasn't the plan to place spatial anchors on AprilTags, and orient the coordinate system to the field and *not* 3d apriltag pose solving?

SeanErn, 2025-12-16:
You have to do 3d pose solving to place the anchors on the tags. If we only do single tag, then their is ambiguity of the direction the tag is facing.

SeanErn, 2025-12-16:
https://docs.wpilib.org/en/stable/docs/software/vision-processing/apriltag/apriltag-intro.html#d-to-3d-ambiguity

blaze-developer, 2025-12-16:
It looked like Juan had a viable product placing AprilTags on the field, and it just needed some tuning. Where did that go?

SeanErn, 2025-12-16:
I'm not trying to be rude, but since this is such a critical feature I am writing it all. By hand. I don't trust AI for this feature. It still exists in this repo, but isn't at the state we want it at. It's being used for some interpretation, but nothing more :)