QuestNav/QuestNav #32: Get AprilTag position and place a spatial anchor

juchong, 2025-03-24:
Detecting an AprilTag is not enough. We need to drop a spatial anchor at each tag location so as to keep track of field changes.

PatMcN7, 2025-08-13:
What will a spatial anchor allow you to do? How will the spatial anchors be used once they are placed?

blaze-developer, 2025-08-14:
Spatial Anchors simply act as reference points in a space, in my understanding, they reflect an unmoving position in space that the Quest can keep track of relative to itself. I believe an example of this is like the Quest's start menu, you can move around a space and it stays in the same physical place. I assume that QuestNav would keep track of similar physical locations and use their relative positions to calibrate the delivered pose into the field coordinate system with the given reference points being AprilTags.

PatMcN7, 2025-09-08:
So will the spatial anchor update it's position as the field drifts? Or will it simply show you that the field has drifted so that you can update it yourself.

closed 2026-10-20 by SeanErn