QuestNav/QuestNav #46: Unity Compiler Errors on Linux

crummyh, 2025-04-01:
I'm trying to open the project on Linux, and have followed all of the instructions in `README.MD` (Will required adjustments, of course). I can open the project, but I get the following compiler errors:

```
Library/PackageCache/com.meta.xr.mrutilitykit/Core/Scripts/MRUKNative.cs(59,8): error CS1029: #error: 'Unsupported platform'
Library/PackageCache/com.meta.xr.sdk.core/Scripts/Editor/OVRManagerEditor.cs(352,79): error CS0103: The name 'manager' does not exist in the current context
Library/PackageCache/com.meta.xr.sdk.core/Scripts/Editor/OVRManagerEditor.cs(353,171): error CS0103: The name 'manager' does not exist in the current context
Library/PackageCache/com.meta.xr.sdk.core/Scripts/Editor/OVRManagerEditor.cs(353,216): error CS0103: The name 'manager' does not exist in the current context
Library/PackageCache/com.meta.xr.sdk.core/Scripts/Editor/OVRManagerEditor.cs(354,169): error CS0103: The name 'manager' does not exist in the current context
Library/PackageCache/com.meta.xr.sdk.core/Scripts/Editor/OVRManagerEditor.cs(354,214): error CS0103: The name 'manager' does not exist in the current context
```

I tried removing `Library/PackageCache`, and all it did was add the following warning:
```
The following asset(s) located in immutable packages were unexpectedly altered. Making changes to immutable packages isn't recommended because the changes can be lost without any warning during Package Manager operations.
  Packages/com.meta.xr.sdk.audio/editor/MetaXRAcousticMaterialMappingEditor.cs
  Packages/com.meta.xr.sdk.audio/Runtime/scripts/MetaXRAcousticMaterialMapping.cs
```

I know that Linux support is not important at all, I was just wondering if anyone had any ideas for solving this. Thank you for this awesome piece of software!

SeanErn, 2025-04-01:
C# is a .NET platform. .NET uses compiled dlls for dependencies. Linux uses another format which it has to be cross compiled for. Meta only supports Windows with there SDKs

crummyh, 2025-04-01:
Found the issue! `manager` is created in a compiler *if* statement:
```
#if UNITY_EDITOR_WIN || UNITY_STANDALONE_WIN || UNITY_EDITOR_OSX || UNITY_STANDALONE_OSX || UNITY_ANDROID
```
Meta dosn't support linux. :confused:

closed 2025-04-01 by crummyh