QuestNav/QuestNav #47: A couple questions.

blaze-developer, 2025-04-04:
I am curious about a couple things.

First of all, how is latency managed in this system, on something like a limelight, a syncronized timestamp is supplied to the robot over NT, which the robot code can use as a timestamp in their pose estimator. Is there a similar way we can get the timestamp of image capture? It seems like something that could be neccesssary.

Next, looking at the code examples, I noticed that when parsing translations from the NT, the X, Y and Z coordinates are in a different order than in standard WPIlib, and in their parsing, the code does this:
```java
private Translation2d getQuestNavTranslation() {
  float[] questnavPosition = questPosition.get();
  return new Translation2d(questnavPosition[2], -questnavPosition[0]);
}
```

[image]

To my understanding, the index orders make sense here, the Z axis gets mapped to the robots X axis, and the X axis is mapped to the robots Y axis. However, why is the Y axis negated? Is the coordinate system in the oculus different than I expect? And if it is, could we have some information on these axises? It would be useful to have documentation detailing the format of all of the NT fields, but I cant find anything of the sort.

Thank you for your wonderful contribution to the FIRST community, and thank you for your response :)

blaze-developer, 2025-04-04:
I thought I could label this as a question, but I dont know how, OOPS

SeanErn, 2025-04-04:
1. Latency is based off the `getTimestamp()` method in the `QuestNav.java` file. This method returns the latency from when the NT table answer was sent from the quest. While not exactly when the measurement was taken, because the Quest tracks at 120hz, the difference is negligible. Only the server time really has to be accounted for
2. We are mapping unity to frc coordinates here. This is just how unity does it. We may switch it to provide in WPILib coordinate frame right off the bat, but for now, this is where it is. I am working on documentation currently, but it isnt ready quite yet.
3. It's a simple command and response. It's not necessarily bad to leave the response codes propagated, but I would suggest against leaving the command codes propegated. The Oculus should (keyword should) only respond or run the action when it changes, but this is relatively untested. Idle is simply saying that the quest has no command.

If you want more help or questions, I'd suggest joining the Discord server. 
https://discord.gg/Sr8Q7SXY

blaze-developer, 2025-04-04:
One more quick question, documentation says that we can reset the questnav to a specific pose, however there doesnt seem to be any communication for that, and the example project doesnt seem to use the supplied pose in the reset, only resetting the navquest to 0, 0, and subtracting the previous pose from it? I dont understand the role of this subtraction if its based on the previous pose and not the new supplied one.

Thanks for your swift responses :3

SeanErn, 2025-04-04:
> One more quick question, documentation says that we can reset the questnav to a specific pose, however there doesnt seem to be any communication for that, and the example project doesnt seem to use the supplied pose in the reset, only resetting the navquest to 0, 0, and subtracting the previous pose from it? I dont understand the role of this subtraction if its based on the previous pose and not the new supplied one.
> 
> Thanks for your swift responses :3 I will be joining the discord server mentioned.

Technically, there is a way to reset the position of the Quest to a specific pose, rather than doing said transform on the robot side. I believe it was forgotten to be put in the example java project. I'll check on this.

closed 2025-04-05 by SeanErn