QuestNav/QuestNav #57: Add robotToQuest to QuestNav.java

ThatChair, 2025-04-24:
Like most teams, we didn't mount our Quest in the center of our robot. Thus, the position we received was offset by a specific amount. We fixed this in our robot code with a robotToQuest translation, and it would be a good addition to QuestNav.java.

SeanErn, 2025-04-24:
I will be updating the documentation to make this more clear!

ThatChair, 2025-04-24:
Most of that makes sense. Why is QuestNav.java solely meant for communication?

SeanErn, 2025-04-25:
QuestNav.java is solely meant for communication because not everyone's robot is the same. We have abstracted out the most critical part of the system---the part that everyone has the same---QuestNav from it. We want to give teams the opportunity to do their own programming too! We still provide great documentation on how to transform the Quest's pose though!

ThatChair, 2025-04-25:
Ah okay. Thanks for expaining!

closed 2025-04-25 by ThatChair