QuestNav/QuestNav #59: No unitypackages for MessagePack

isubsmith, 2025-04-30:
I'm trying to build the APK file. The MessagePack-CSharp repo doesn't have a .unitypackage file available. 

I've fought this battle before early in the season, but don't remember where I eventually got a unitypackage.

The docs say:
Download the latest release with a .unitypackage extension from [the MessagePack-CSharp GitHub repository](https://github.com/MessagePack-CSharp/MessagePack-CSharp/releases)

However there are just .nupkg and .snupkg files there. If I use something like nuget to install the .nupkg files, will they actually be found/used by Unity?

Thanks

juchong, 2025-04-30:
Take a look at earlier Messenger pack releases. They sometimes don't build the .unitypack file with every release for some reason.

isubsmith, 2025-04-30:
Ok, finally found one in their 2.5.192 tag. 
Thanks.

closed 2025-04-30 by isubsmith