QuestNav/QuestNav #61: No Adapter Found

NautRobot, 2025-05-05:
When I follow the setup directions and connect everything, the Quest Message just reads, "No Adapter Found" and some stuff about reconnecting in three seconds, but it never does. Is there any current working fixes?

SeanErn, 2025-05-05:
Is the ethernet adapter plugged in and connected to a wired network?

NautRobot, 2025-05-05:
Yes, the adapter lights are blinking. It's plugged in to an Ethernet Switch, should I move it to the VH-109 aux ports? (one of them)

SeanErn, 2025-05-05:
The quest has to have an IP so if your switch is managed or not properly assigning IP addresses it will not work. Please also check that WIFI is completely disabled

NautRobot, 2025-05-05:
WiFi is completely disabled, is there any instructions to check if the switch is managed or to make sure its properly assigning IP addresses?

NautRobot, 2025-05-07:
It appears to work, (sorry for forgetting to update), is this meant to be solved this way?

SeanErn, 2025-05-07:
What way do you mean?

NautRobot, 2025-05-12:
Are you supposed to have it plugged into an Aux port or is it okay if its plugged into an ethernet switch?