QuestNav/QuestNav #63: Automatic Startup

amorygalili, 2025-05-07:
Would be great to have for the reasons already mentioned in the documentation:

[image]

Starting the app currently requires me to open the app manually while wearing the headset or launch it with an ADB command using the SideQuest. Both are inconvenient when the quest is mounted to the robot and difficult to access.

SeanErn, 2025-05-07:
Of course! We have been looking for a way to do it. If you find one, make a PR! 

This will probably be lower on the priority list, but still on there!

amorygalili, 2025-05-08:
Have it partially working. It starts up automatically when powered on, but the rest of the quest 3 UI becomes inaccessible unless you press the power button twice. I think the app is in fullscreen mode and this is the only way to make it lose focus. Still trying to figure out why this is happening.

Also haven't looked into restarting the app yet after unexpected restarts.

agasser, 2025-07-14:
I created #79 to automatically launch on boot. It works the way I expect it to, but feedback is welcome!

closed 2025-07-17 by SeanErn