QuestNav/QuestNav #64: Add simulation support

amorygalili, 2025-05-11:
Add simulation support while quest is connected to laptop through USB. This can be achieved using a reverse proxy by running this adb command in SideQuest:

```bash
adb reverse tcp:5810 tcp:5810
```

A couple additional modifications need to be made including setting the list of candidate NT4 addresses to localhost/127.0.0.1 and adding a "Connect to Sim" button, but otherwise this is a pretty simple feature to implement.

SeanErn, 2025-05-11:
Feel free to open a pull request to add this feature! We did have a pull request for the same thing, but we did not merge it because of issues with power leakage and won't be supporting USB as an official way to connect to a RoboRIO in a production environment. A connect to sim button would solve this! Just make sure you only add 127.0.0.1 to the candidates if the connect to sim button is pressed please.

amorygalili, 2025-05-11:
Just added a PR. Will change the candidate list to only use 127.0.0.1. Don't think adding localhost was necessary for it to connect.