QuestNav/QuestNav #81: Leverage NT4 queue features in robot code

agasser, 2025-07-16:
NT4 subscribers can read data using a queue. By default, the queue only saves the last update (unless `PubSubOption.sendAll(true)` or `PubSubOption.pollStorage(x)` is set. The vendordep could return a list of all of the updates (up to a reasonable limit) since the last time the method was called. PhotonVision's [`getAllUnreadResults()`](https://javadocs.photonvision.org/release/org/photonvision/PhotonCamera.html#getAllUnreadResults()) does this, for example.

This approach can make integration easier for teams. Today, calling `getPose()` in robot code will return an invalid pose (0, 0, 0°) if it's called when there is no new data. Teams need to check for this invalid pose (a common oversight we see with naive Limelight integrations). Teams are also advised to check the `isConnected()` method before reading updates. If the vendordep only returned updates since the last call, it wouldn't matter if QuestNav was disconnected, it just wouldn't have any updates so an empty list would be returned.

Teams would also get a full set of updates they can use for logging or pose estimation, even if more than one update is published between periods.

I'm happy to work on a PR for this. I think it would make sense for this to be dependent on #80, so the timestamp from all of the pose estimates can be returned.

SeanErn, 2025-07-17:
This sounds great! Especially because the Quest-side loop runs every 8.33ms instead of the 20ms that the robot uses. I'd love to see it!

agasser, 2025-07-17:
Yes, exactly. This way no frames are dropped even though the robot loop runs slower.

Side note, I think the Quest-side loop runs ever 10ms (100Hz) now, unless I'm misunderstanding. See https://github.com/STMARobotics/QuestNav/blob/main/unity/Assets/QuestNav/Core/QuestNav.cs#L157

SeanErn, 2025-07-17:
Oops yeah my bad. We could probably bump that up to 120hz when this gets merged

PR #84 (2025-07-22): Get pose and timestamp from the same frame, and use NT4 queue features in questnavlib
All data that's sent in the pose frame is retrieved as a unit. The caller will get all of the unread frames (up to a max of 20) when they call `getAllUnreadPoseFrames`. ~Methods that retrieved individual values from the frame were removed to avoid popping the frame from the queue (and to simplify the interface.)~

I also changed the methods that might not have data from the Quest to return Optionals instead of default values.

Resolves #80 
Resolves #81